dhansel/ArduinoFDC #1: Extension for HD possible?

Scotty42, 2021-04-06:
I was wondering if hardcoded gaps, sectors, etc or the required data rate of 500 kbit vs. 250 kbit is the actual limiting factor that blocks high density (1,2 MB + 1,44 MB) support?

dhansel, 2021-04-13:
The limiting factor is the data rate. At 500 kbit, a new data bit
arrives from the drive every 2 microseconds. At 16MHz clock frequency
that's only 32 processor cycles. The current read/write implementation
in C using timers is not fast enough for that.

I am considering rewriting the read/write code in assembly which
would make it possible ensure correct timing by adding up the processor
cycles needed for each instruction. That should be fast enough.

Scotty42, 2021-04-13:
Was expecting this. Thanks for clarification.

closed 2021-04-13 by Scotty42

dhansel, 2021-04-18:
As expected, re-writing the read/write routines in assembly solved the timing issues
for HD disks. Timing is close but works as far as I can tell from my testing.

The controller now supports 3.5" and 5.25" DD and HD disks.

Scotty42, 2021-04-19:
That's awesome! I'll do some tests and report back.

Scotty42, 2021-04-20:
Did connect 3,5" HD as A and 5,25" HD as B on a standard floppy cable with twist on A. Host end of the cable connected to some pinheaders and an Arduino Uno board. Although I can verify Arduino pins are connected to the floppy ports according to README table, I fail to read/write/format anything. I can select s 0/1 and turn motor 0/1. Any other operations start motors, light up LEDs but nothing actually happens. Head does not step. Do you have any idea to further analyze and debug?

dhansel, 2021-04-21:
I just tried it again here, uploaded the code from GIT to the Arduino and re-connected my drive
following the wiring from the instructions. Works fine for me in that configuration (tested two different
drives). So at least the wiring/code is not completely broken :)

Do you get any error message when trying to read a sector ("r 0 1")? If so, what is the error number?
If not, does the disk LED come on when trying to read? Does it stay on and the process just hangs
or does the LED go off again? Is there any out in the terminal? try un-commenting the "#define DEBUG"
statement in ArduinoFDC.cpp and uploading again. Does that show any more information?

Try and add the following code at the end of the AduinoFDCClass::begin() function in ArduinoFDC.cpp:
```
  digitalWriteOC(PIN_SELECTA, LOW);
  step_to_track0();
  step_tracks(m_driveType[0], 20);
  step_to_track0();
  digitalWriteOC(PIN_SELECTA, HIGH);

```
After uploading you should hear the drive head step (to track 0, then track 20, then back to 0).
If that does not happen then most likely something is wrong with the wiring. Maybe the GND
signal is not properly connected? I've seen drives/cables where not all odd-numbered pins are
actually connected to ground.

Generally the code is not supposed to hang. Even with no drive connected at all it should at least
display an error message. If the code is hanging when reading, try putting a `return S_OK;` 
statement at the beginning of function "read_data" in ArduinoFDC.cpp
That will skip all read attempts completely. Does that still hang? If not then that may indicate
that the code is getting stuck in the assembly section(s).

Scotty42, 2021-04-21:
I wasn't doubting your tests :). Really appreciate above tipps 👍 
As almost always after reaching out to someone, problem solves itself. No, not on itself, but I can trace it back to a faulty head DIR line in the cable and some weird behavior with GND connections. I had:

- all connected 
- one
- two as Arduino Uno offers 2 GND pins 

And after randomly (re)selecting 2 GND pins finally a 3,5" Floppy was accessible. Haven't yet tried 5,25" again, but I'd say it is save to assume it will work now, too.

dhansel, 2021-04-21:
Glad to hear it's working! When trying the 5.25" drive make sure to set the correct drive type, either by changing
the parameters to the ArduinoFDC.begin() function or by using the 't' command.

Scotty42, 2021-04-21:
I can confirm 5,25" HD drive working with HD floppy and mode t2. 
I fail to r/w/f a DD floppy though. r/w yields error No sync marks found! and f leads to some undefined operation for minutes that ends without error but does not do anything.

dhansel, 2021-04-21:
Does the head step while trying to format the DD disk The format process does not read anything back to see whether the format worked. It just formats a track and moves on to the next. So 'f' returning OK doesn't really mean much.

I assume you mean you're having problems with a 5.25" DD disk. Which disk type are you setting? Since you have a 5.25DD 
disk in a HD drive you need type 1 (5.25DDonHD). Also make sure to read the README section about the DENSITY 
pin. Also note that you need a true 5.25" DD disk. You can try and format a HD disk as DD but that does not always work.

Thanks for the correction on the "t" command help. I'll fix that. The proper order is:
0=5.25DD / 1=5.25DDonHD / 2=5.25HD / 3=3.5DD / 4=3.5HD

Scotty42, 2021-04-21:
yes, I was testing a real DS/DD floppy with 48TPI and setting t := 1. I'm fortunately mature enough for having used all sorts of floppies in real live, not only nowadays for retro reminiscent feelings or curiosity :). 
The floppies tested had been formatted in one of my classic PC gear and actually even on the drive I momentarily use for this Arduino project. 
Head steps from track 0..max during format. Resets back to 0 to write DOS sector but does not write anything to the disk.

dhansel, 2021-04-21:
I think the problem is that the density pin mode doesn't get automatically switched when you change
drive types via the 't' command. Try replacing `setDensityPin();` with `setDensityPinMode((type==DT_3_HD || type==DT_3_DD) ? DP_DISCONNECT : DP_OUTPUT_LOW_FOR_DD);` in the ArduinoFDCClass::setDriveType() function.
Note: I can't try this myself right now so hopefully I didn't make any typos.

Scotty42, 2021-04-21:
will give it a shot. Figured already between 
`#if defined(PIN_DENSITY)
  // by default: 3.5"  drives do not use DENSITY pin (disconnect)
  //             5.25" drives expect DENSITY to be LOW for low density
  m_densityPinMode[0] = (m_driveType[0]==DT_3_HD || m_driveType[0]==DT_3_DD) ? DP_DISCONNECT : DP_OUTPUT_LOW_FOR_DD;
  m_densityPinMode[1] = (m_driveType[1]==DT_3_HD || m_driveType[1]==DT_3_DD) ? DP_DISCONNECT : DP_OUTPUT_LOW_FOR_DD;
  digitalWrite(PIN_DENSITY,   LOW);
  pinMode(PIN_DENSITY,   INPUT);
#endif`
initialization and later 
`void ArduinoFDCClass::setDensityPin()
{
#if defined(PIN_DENSITY)
  byte isHD = m_driveType[m_currentDrive]==DT_3_HD || m_driveType[m_currentDrive]==DT_5_HD;
  switch( m_densityPinMode[m_currentDrive] )
    {
    case DP_OUTPUT_LOW_FOR_DD:
      digitalWriteOC(PIN_DENSITY, isHD ? HIGH : LOW);
      break;
      
    case DP_OUTPUT_LOW_FOR_HD:
      digitalWriteOC(PIN_DENSITY, isHD ? LOW : HIGH);
      break;

    default:
      digitalWriteOC(PIN_DENSITY, HIGH);
    }
#endif 
}`
in fact nothing changed by type setting. So default is LOW. For my TEAC 55GFR on the test bench that is HD so the reason it works for HD per default is clear

dhansel, 2021-04-21:
I'm also using a TEAC 55GFR here, so chances are good that you should be able to get it to work.
I'm not following 100% what you mean by "default is LOW". If the DENSITY pin is not connected then
it defaults to HIGH (via the pullup resistor). In its default configuration, the 55GFR expects the DENSITY
pin to be HIGH (or unconnected) for high density and LOW (pulled to GND) for low density.
You can find a user manual for the 55GFR [here](https://github.com/dhansel/Altair8800-IOBus/blob/main/04-disk-controller/doc/TEAC%20FD-55GFR.pdf).

Scotty42, 2021-04-21:
was a typo. Already corrected in the comment. default is HIGH.
With the change t1 => LOW and t2 => HIGH on PIN2. Still no positive result for DD disks. 
HD works flawlessly and still works after the change.

```
Command: r 0,1,0
Reading track 0 sector 1 side 0
Error: Sector header not found!
```

or 

```
Command: r 0,1,0
Reading track 0 sector 1 side 0
Error: No sync marks found!
```

dhansel, 2021-04-21:
I understand now, thanks.. We're both saying the same thing. Weird that it still won't work. I'll give it another
try here later tonight.

dhansel, 2021-04-22:
I think I know what the issue is: the drive rotation rate. Original 5.25" DD drives rotated at 300 RPM. When HD drives came around, the rate was  increased to 360 RPM. When you have a DD disk in an HD drive then you have two possibilities:
1. Use the slow 300 RPM rotation rate and a data rate of 250 kbps (just like an original DD drive).
2. Use the fast 360 RPM rotation rate and an increased data rate of 300 kbps.

Both will cause the disk to rotate under the head for the same amount of physical space for each data bit read/written.

Regular floppy disk controllers generally use option (2). So when you have the drive connected to your computer, the controller there will expect the drive to always rotate at 360 RPM and adjust its data rate. to 300 kbps.

My code currently is set up to use a data rate of 250kbps for DD, requiring the drive to switch to 300 RPM in DD mode. I remember now that I set up my drive to do that (when I first started working with 5.25" DD disks in the 55GFR).

However, it should be pretty simple to fix the code: Just change the "32" to "27" in line 55 (the bitLength field of the drive geometry structure for the DT_5_DDonHD drive type).  That will change the bit rate for that type. Note that 27 is not quite the correct value, it should be 26.6666... - but processor cycles are atomic so I can't make it more precise. The timing will be off by 1.25% which should be ok - as far as I know the drive rotation speed itself has a tolerance of about 3%.

I've done some tests here with that timing on DD and have not had any problems reading or writing. 

Alternatively, you can re-configure your 55GFR to switch to 300 RPM in DD mode by removing the "II" strap (which is set by default) and moving it to "IS". That will cause the drive to slow its rotation to 300 RPM when the "density" signal goes low. You can find more info on that on page 133 of the 55GFR manual I linked to before.

Scotty42, 2021-04-22:
I was reviewing gaps and geometry yesterday evening with data from [http://www.moria.de/~michael/floppy/floppy.ps](url) but couldn't find an issue; I had been expecting timing issues and thought gaps would directly relate. 
Anyway, changing the bit length did the trick. DD in HD drive works 👍

dhansel, 2021-04-23:
Great to hear it's all working now. Let me know if you find more problems.
FYI - I have applied the fixes for the 5.25" DD disks in HD drives to the repository. For the rotation rate issue, the code 
now measures the rotation rate and automatically uses the corresponding data rate.

Scotty42, 2021-04-23:
Actually I found two things yesterday. Firstly parameter order for read and write seem not identical t/s/h vs t/h/s. Secondly a DD disk is not really writeable with a floppy controller and another drive. Tested with a 386. Dir listing works but as soon as some data is written, write errors pop up. Timing seems to remain critical. But that was already a problem in the days when DD written on a HD drive were exchanged

dhansel, 2021-04-23:
I took a look at the code (and tried reading/writing here) and as far as I can see both read ('r') and write ('w') commands use t/h/s
for the parameter order. What makes you think the order is different?

Regarding writing DD in another drive: Are you saying that a DD disk formatted using ArduinoFDC is not readable on a different controller and drive? Is the other drive a true DD drive or an HD drive? It is a general problem that a DD disk written (or formatted) in a HD drive can often not properly be read in a DD drive. That's because the HD head is smaller than a DD head and the magnetization can be too weak for the DD head to consistently detect.  Since writing to a disk also involves reading (to find the sector), the problem may actually be a read error due to this effect. If this is the cause then I don't think anything can be done - it's a hardware issue. As you say, writing to a DD disk in a HD drive has always caused problems.

Scotty42, 2021-04-24:
Nevermind, it was late. Functions are consistent in their respective context. But not in-between for easy comparison. Like writing boot sector                   `ArduinoFDC.writeSector(0, 0, 1, data, false);` and reading it according to serial menu r 0,1,0. 

I know the technical limitations of HD and DD drives. No worries. Therefore I meant exactly the scenario of reading and writing a formatted DD floppy, formatted in mentioned HD drive, and then writing to it in a real computer's HD drive. Which I think is either Panasonic brand or Teac, too. That doesn't work without errors.

dhansel, 2021-04-25:
I did a bunch of testing this weekend but was not able to reproduce the problem with writing DD disks. All 5.25" DD disks formatted by ArduinoFDC were  always readable and writable on a Win98/DOS machine. However, I only have one 5.25" drive so I can't exactly reproduce your setup.

Two things come to mind that you may want to try (if you haven't already).

1. Maybe the problem is with the FAT file system initialized by ArduinoFDC rather than the low-level format. You could try formatting a disk with ArduinoFDC, then format it again on the DOS computer with a "/q" argument to ONLY reset the filesystem. If writing works after that then that would indicate that the file system (rather than the low-level format) is the problem.
2. It is possible (though unlikely) that the 1.25% timing difference with the 300kbps data rate is causing a problem. Again, I was not able to reproduce that but who knows. You could try changing the straps on your 55GFR to have it switch to 300RPM for DD mode. The latest versions of ArduinoFDC automatically determine the data rate and for 300RPM will use the precise 250kbps data rate instead.

Anyways, let me know if you find anything else.

BTW - I just added support for directly reading/writing files on a FAT file system as well as the possibility to transfer full disk images and single files via XModem.

Scotty42, 2021-04-27:
Actually found some time now to re-test using the latests code. Impressive development in-between the two versions. 
I was going with your scenario 2, because I used two different drives, both with tolerances. 

Good news is, with latest code, I could not reproduce the error. A 5,25" DD could be formatted with Arduino, read and written by DOS (tried to sys it which worked flawlessly). Then I changed drives back, wrote a test file with ArduoDOS, which I then edited with Norton Commander on that DOS machine.