dhansel/ArduinoFDC #2: USB mass storage Uniform Floppy Interface device class

VasylSamoilov, 2021-05-06:
Can it be used as usb floppy controller (implement UMS UFI device class)? There is no USB controllers for 5.25, and never were, and this may be a chance to connect 5.25 floppy to modern PC via USB :)

dhansel, 2021-05-06:
Neither the Arduino UNO nor the Mega have the capability to act as anything other than a serial-over-USB device. As far as I know all Arduinos or other devices that are capable of more generic USB handling use a different processor architecture so would require rewriting large portions of the code (the critical parts are implemented directly in processor-specific assembly language).

So I don't think this can be converted directly into a usb floppy controller. You could implement the USB communication part on a different device and then talk to the Arduino via serial the serial connection to read/write data on the disk.

closed 2021-05-07 by dhansel

VasylSamoilov, 2021-05-07:
Most unofficial Uno and Mega boards probably can't, as they use CH340G as USB-to-Serial converter. But those Arduino that has a USB connection in the main MCU, like Leonardo, Micro, Pro Micro, Due, Zero and LilyPad USB can probably use https://github.com/Palatis/Arduino-Lufa (port of Lightweight USB Framework for AVRs), which includes USB mass storage class.

dhansel, 2021-05-07:
True but as I said the ones that have USB support in their main MCU are using a different MCU so I don't know
if my code will run on them since it is highly optimized for the Atmega328p. Apart from the Due (on which it 
definitely doesn't work) I don't have any of them so I can't test it here.

Pablo2048, 2021-05-07:
Maybe we can use Arduino Micro Pro - uses Atmega32Ux - AVR core, compatible with Atmega328p AFAIK...

dhansel, 2021-05-07:
I checked out the Micro Pro/Atmega32Ux and that does look promising. I'll get one and check out
if the ArduinoFDC library works on it. I'll leave figuring out the USB protocol to someone else though...