dhansel/ArduinoFDC #3: Compile error

rtrimbler, 2021-05-06:
Not sure how you are building this project, but with Arduino IDE 1.8.13, I could not get the code to compile.
Getting an error here:

file: ArduinoFDC.ino
line 227: void print_ff_error(FRESULT fr)
error: FRESULT not declared in this scope

With enough searching the web, I finally figured out it doesn't deal well with the typedef enum as a variable type. Had to add a forward reference just ahead of this line and all is well:

void print_ff_error(FRESULT fr);

Other than that, it seems to be working OK. I've only tried it with a Teac FD-235 hf drive and a 3.5" DD disk. I can format & r/w in the monitor; re-formatted in ArduDOS & created a couple of directories. Oddly, putting this disk into a USB floppy drive on Win10 still reads the original directory (albeit very hesitantly.)

Great project, thanks for sharing! My goal is to try and read some old 1.2M (HD?) 5.25" floppies. I do have a teac fd-55gfr, but can't find the ribbon cable with the card edge connector for it.

-- Reid

Jorin-Post, 2021-05-06:
Hello,

I'm not sure but try to change void print_ff_error(FRESULT fr) to void print_ff_error(&FRESULT fr)
If you want to sent data back to the previous function & will change the incoming data to the new data if the data is finished.
success not guaranteed ;p

-- jorin

rtrimbler, 2021-05-06:
Success certainly isn't guaranteed with that - the syntax is wrong. void print_ff_error(FRESULT &fr) would in fact pass fr by reference, but that's not what's needed here. print_ff_error has no need to modify fr and the calling function wouldn't expect it to. I already had success with my suggestion; just not sure why it never showed up for you. Seems like this is a known problem with how the arduino IDE builds the .cpp file it passes off to the compiler.

closed 2021-05-06 by rtrimbler

dhansel, 2021-05-06:
Very strange - I use the same Arduino IDE version (1.8.13) and have no problem compiling. I tried re-downloading everything from GitHub but that compiled fine too.

What platform are you on (Windows/Linux/Mac)? I am on Windows. Maybe that makes a difference.

Can you let me know where exactly (line number) you inserted the forward declaration? Then I can just put it there for safety (can't hurt). But since I can't reproduce the problem here I need to know exactly where to put it,

Thanks for reporting this!

rtrimbler, 2021-05-07:
I am using Win10 as well, so that's pretty strange behavior. I think arduino uses a version of GCC though, and I don't have anything else loaded that may have updated that.

I literally put the forward declaration 1 line before the formal declaration. So on the latest code I found on github, the forward decl is on line 227 and the actual function is now on line 229 (I like white-space.)

dhansel, 2021-05-07:
Ok I added the declaration. Hopefully it will save someone from having to deal with the compile error you encountered.
Thanks again!

rtrimbler, 2021-05-07:
Looks good! 
2nd problem - I've tried several disks, both DD and HD (all 3.5") that read fine on a windows/usb floppy, but this project won't read any sector/head for tracks 0 & 1. 
It reports this (with #DEBUG enabled)
Command: r 0,1,0
Reading track 0 sector 1 side 0
Unable to find header!
Unable to find header!
Error: Sector header not found!

Tracks 2-79 read fine. Of course, without the first couple of tracks ArduDOS can't see any directory. Any ideas?

rtrimbler, 2021-05-07:
Sorry for the delay.  My code was somewhat different than the base b/c I changed the pins to make wiring easier on myself. Spent the morning trying to figure out why your debug info wasn't appearing; then realized I had patched incorrectly. Here is the output. I read a sector from track 2 first (which works) then from track 0.

```
Drive A: 3.5" HD
Drive B: 3.5" HD

Command: r 2,1,0
Reading track 2 sector 1 side 0
HFE02001202+
HFB82023930-
0000: 30 30 30 20 49 6E 74 65  6C 20 43 6F 72 70 6F 72   000 Inte l Corpor
0010: 61 74 69 6F 6E 0D 0A 0D  0A 3B 20 2A 2A 2A 2A 2A   ation... .; *****
0020: 2A 2A 2A 2A 2A 2A 2A 2A  2A 2A 2A 2A 2A 2A 2A 2A   ******** ********
0030: 2A 2A 2A 2A 2A 2A 2A 2A  2A 2A 2A 2A 2A 2A 2A 2A   ******** ********
0040: 2A 2A 2A 2A 2A 2A 2A 2A  2A 2A 2A 2A 2A 2A 2A 2A   ******** ********
0050: 2A 2A 2A 2A 2A 2A 2A 0D  0A 3B 20 49 6E 74 65 6C   *******. .; Intel
0060: 20 69 73 20 6D 61 6B 69  6E 67 20 6E 6F 20 63 6C    is maki ng no cl
0070: 61 69 6D 73 20 6F 66 20  75 73 61 62 69 6C 69 74   aims of  usabilit
0080: 79 2C 20 65 66 66 69 63  61 63 79 20 6F 72 20 0D   y, effic acy or .
0090: 0A 3B 20 77 61 72 72 61  6E 74 79 2E 20 20 0D 0A   .; warra nty.  ..
00A0: 3B 20 2A 2A 2A 2A 2A 2A  2A 2A 2A 2A 2A 2A 2A 2A   ; ****** ********
00B0: 2A 2A 2A 2A 2A 2A 2A 2A  2A 2A 2A 2A 2A 2A 2A 2A   ******** ********
00C0: 2A 2A 2A 2A 2A 2A 2A 2A  2A 2A 2A 2A 2A 2A 2A 2A   ******** ********
00D0: 2A 2A 2A 2A 2A 2A 2A 2A  2A 2A 2A 2A 2A 2A 0D 0A   ******** ******..
00E0: 3B 20 49 6E 66 6F 72 6D  61 74 69 6F 6E 20 69 6E   ; Inform ation in
00F0: 20 74 68 69 73 20 64 6F  63 75 6D 65 6E 74 20 69    this do cument i
0100: 73 20 70 72 6F 76 69 64  65 64 20 69 6E 20 63 6F   s provid ed in co
0110: 6E 6E 65 63 74 69 6F 6E  20 77 69 74 68 20 0D 0A   nnection  with ..
0120: 3B 20 49 6E 74 65 6C 20  70 72 6F 64 75 63 74 73   ; Intel  products
0130: 2E 20 20 4E 6F 20 6C 69  63 65 6E 73 65 2C 20 65   .  No li cense, e
0140: 78 70 72 65 73 73 20 6F  72 20 69 6D 70 6C 69 65   xpress o r implie
0150: 64 2C 20 62 79 20 65 73  74 6F 70 70 65 6C 20 0D   d, by es toppel .
0160: 0A 3B 20 6F 72 20 6F 74  68 65 72 77 69 73 65 2C   .; or ot herwise,
0170: 20 74 6F 20 61 6E 79 20  69 6E 74 65 6C 6C 65 63    to any  intellec
0180: 74 75 61 6C 20 70 72 6F  70 65 72 74 79 20 72 69   tual pro perty ri
0190: 67 68 74 73 20 69 73 20  67 72 61 6E 74 65 64 20   ghts is  granted 
01A0: 0D 0A 3B 20 62 79 20 74  68 69 73 20 64 6F 63 75   ..; by t his docu
01B0: 6D 65 6E 74 2E 20 20 49  6E 74 65 6C 20 61 73 73   ment.  I ntel ass
01C0: 75 6D 65 73 20 6E 6F 20  6C 69 61 62 69 6C 69 74   umes no  liabilit
01D0: 79 20 77 68 61 74 73 6F  65 76 65 72 2C 20 0D 0A   y whatso ever, ..
01E0: 3B 20 61 6E 64 20 49 6E  74 65 6C 20 64 69 73 63   ; and In tel disc
01F0: 6C 61 69 6D 73 20 61 6E  79 20 65 78 70 72 65 73   laims an y expres

Command: r 0,1,0
Reading track 0 sector 1 side 0
HFE02000702+
HFB005F0032-
HFE02000802+
HFB311F301D-
HFE02000902+
HFB003081C3-
HFE02000A02+
HFE02000B02+
HFE02000C02+
HFE02000D02+
HFE02000E02+
HFE02000F02+
HFE02001002+
HFE02001102+
HFE02001202+
HFE02000D02+
HFB0E566572-
HFE02000E02+
HFB5A7A657B-
HFE02000F02+
HFB20436F72-
HFE02001002+
HFE02001102+
HFE02001202+
HFE02000102+
HFE02000202+
HFE02000302+
HFE02000402+
HFE02000502+
HFE02000602+
HFE02000702+
HFE02000802+
HFE02000902+
HFE02000A02+
HFE02000B02+
HFE02000C02+
HFE02000D02+
HFE02000E02+
HFE02000F02+
HFE02001002+
HFE02001102+
HFE02001202+
HFE02000102+
HFE02000202+
HFE02000302+
HFE02000402+
HFE02000502+
HFE02000602+
HFE02000702+
HFE02000802+
HFE02000902+
HFE02000A02+
HFE02000B02+
HFE02000C02+
HFE02000D02+
HFE02000E02+
HFE02000F02+
HFE02001002+
HFE02001102+
HFE02001202+
HFE02000102+
HFE02000202+
HFE02000302+
HFE02000402+
HFE02000502+
Unable to find header!
HFB20436F72-
HFE02001002+
HFB01A381A8-
HFE02001102+
HFB303F0603-
HFE02001202+
HFE02000102+
HFE02000202+
HFE02000302+
HFE02000402+
HFE02000502+
HFE02000602+
HFE02000702+
HFE02000802+
HFE02000902+
HFE02000A02+
HFE02000B02+
HFE02000C02+
HFE02000D02+
HFE02000E02+
HFE02000F02+
HFE02001002+
HFE02001102+
HFE02001202+
HFE02000102+
HFE02000202+
HFE02000302+
HFE02000402+
HFE02000502+
HFE02000602+
HFE02000702+
HFE02000802+
HFE02000902+
HFE02000A02+
HFE02000B02+
HFE02000C02+
HFE02000D02+
HFE02000E02+
HFE02000F02+
HFE02001002+
HFE02001102+
HFE02001202+
HFE02000102+
HFE02000202+
HFE02000302+
HFE02000402+
HFE02000502+
HFE02000602+
HFE02000702+
HFE02000802+
Unable to find header!
Error: Sector header not found!

Command: 
```

dhansel, 2021-05-07:
Ok so the disk reads just fine but even after trying to step to track 0, the sector headers that are 
being read still say that the drive is on track 2.

This looks to me like your drive is out of alignment and can't reach tracks 0 or 1, i.e. the "track 0"
signal goes LOW even though the drive head has not reached track 0 yet. 

Do you have a different drive that you could test with?

rtrimbler, 2021-05-07:
Arrrrgggghhh. It's the hardware.  I just swapped in a different 3.5" floppy (same model Teac, but a different generation it appears) and it works **fine**. Reads all sectors; Ardudos lists a directory & everything. Curiously, the free space report is different: ArduDOS: `1373696 bytes free`, Win10: `1,372,672 bytes free`

rtrimbler, 2021-05-07:
I did find a very small error when wandering thru the code:
ArduinoFDC.cpp, line 154 says
`static struct DriveGeometryStruct geometry[6] =`
pretty sure that should be `[5]`
Doesn't really hurt anything as-is, but the compiler won't notice if you wander off the end.

dhansel, 2021-05-07:
Thanks for the correction. I fixed it in the repository.