dhansel/ArduinoFDC #5: FormatDisk hangs forever

daid, 2022-05-11:
I'm encountering this issue where the formatDisk function hangs forever, and never returns from the first `format_track()` call.

It's quite a "rotten" disk that I'm using. It read track 40 if this disk. But when I tried a `formatDisk(buffer, 0, 0)` it just hangs and keeps spinning the disk forever. Reading track 0 doesn't work, which I figured it might be damaged beyond repair or just lost it's formatting. But I wouldn't expect a bad disk to cause the format to hang forever.

I'm not sure where it can hang. I know the `wait_for_index()` finishes: <https://github.com/dhansel/ArduinoFDC/blob/main/ArduinoFDC.cpp#L740> but somewhere after that it just hangs.

I'm using different pinout from default on an arduino nano. I can read/write other disks just fine. Didn't try format on any other disk, dont want to screw up any other disks that I have.

dhansel, 2022-05-11:
Did you maybe change the pin for the INDEX hole? If so you also have to update the IDXPORT and IDXBIT settings here: https://github.com/dhansel/ArduinoFDC/blob/cd3b1c0417d39498b1bb882b5a2bcb5ef93bb41c/ArduinoFDC.cpp#L51-L52

After writing the track data during format the code keeps writing filler bytes until it sees the index hole again. If you changed the INDEX pin without updating IDXPORT/IDXBIT then the assembly code will just wait forever for the index hole.

daid, 2022-05-12:
Ah, yes. I did change the index pin, but and I changed the IDXPORT, but now I see that I changed it wrong. I changed it to PC5, but made IDXPORT point to the `PORTC` register instead of the `PINC` register (most likely because is named IDX**PORT** )

Stupid mistake from my side. Thanks!

(I think there can be some way to pass these from C to assembly without needing to double define them in ASM and C, I might investigate and make a pull request)

closed 2022-05-12 by daid