dhansel/ArduinoFDC #6: HD Floppy doesn't work without formating it in low leve disk mode

thatsitipl, 2022-05-13:
I have some old HD Floppys but when i try to run the r command it only returns something like:
Reading track 0 sector 1 side 0 => Error: No sync marks found!
After i reformat it works. 

Any Help is appreciated
Best regards

dhansel, 2022-05-13:
Do you know for sure that the disks are properly formatted and readable? It is certainly possible that the magnetization on the disks has degraded over time to a point where they cannot be read anymore.  Formatting the disk then restores it to a readable (but empty) state.

If the old content of the disks is important to you then I would recommend a project like [Greaseweazle](https://decromancer.ooo/greaseweazle) or [Fluxengine](http://cowlark.com/fluxengine/index.html). These projects extract the raw flux data from the disk and use a PC (much more powerful than an Arduino) to decode it. They are much better at error correction than my ArduinoFDC project. Therefore they are much better if the goal is to read important content of old degraded floppy disks. But they only work in conjunction with a PC.

ArduinoFDC is meant for using floppy disks as a storage medium (reading and writing) using only an Arduino. It can also read old disks (I have used it successfully on 20 year old disks) but only if they have not degraded too much.

If the disks work fine after formatting then I think your setup is good and it is just the old age and degradation of the disks that is causing the problem.

thatsitipl, 2022-05-14:
Thank you very much for your response im going to check on the other projects

closed 2022-05-14 by thatsitipl