dhansel/ArduinoFDC #7: Power connection?

zechmer, 2022-07-13:
I did not find a description of how to power the floppy drive with 12V and 5V power connections (I know that you can power it with ATX etc.). In the picture it seems that USB is used for power (aren't the 12V necessary?). Can you please add such a description? What did you use?

mchro, 2022-07-13:
@zechmer Connecting only 5V power should be enough, if not for all floppy drives then for almost all.
Pinout is pretty simple:
* Red: 5V
* Black: GND

https://en.wikipedia.org/wiki/Berg_connector

dhansel, 2022-07-13:
As far as I know 3.5" floppy drives only need 5V power but 5" drives do need 5V and 12V power. In the picture the black cable going to the left is the power cable, coming from an external hard drive power supply ([something like this](https://www.amazon.com/SATA-Adapter-Converter-Cable-Drive/dp/B01HO07GNW)) with an adapter for the smaller 3.5" drive connector.

You should be able to power a 3.5" drive directly from USB, i.e. connect the 5V and GND pins from the Arduino to the 5V and GND pins on the [floppy power connector](https://en.wikipedia.org/wiki/Berg_connector). I have done that before. Use a good USB cable from the PC to the Arduino, otherwise you will get too much voltage drop over the cable and things won't work correctly.

dhansel, 2022-07-13:
I have added a section about the power supply in the description. Thanks for the suggestion!

closed 2022-07-13 by dhansel