dhansel/ArduinoFDC #9: Constantly getting Error #1: Low-level disk error

ACrazyTown, 2023-03-06:
I'm trying to read a 3.5 HD floppy disk, however I constantly get low-level disk errors in ArduDOS while trying to run `dir`.

I'm pretty sure I connected everything correctly as it can detect if a drive is inserted or not, but I don't know why I keep getting the low-level disk errors.
I don't know what could be the problem, it could be the floppy drive, maybe even the disk. Does anyone know of some troubleshooting steps I could try to pinpoint where the issue is?

dhansel, 2023-03-06:
On the ArduDOS command line, type "monitor" which will get you into the low-level disk monitor. 
The disk drive motor should automatically turn on at that point. Check if it's running.
If not, type "s 1" to select drive B (maybe you have the drive wired as drive B). Does the motor run now?
If not then check your wiring. Make sure the drive has power.

If the motor runs then type "r 0,0,0" which should read the first sector on the disk. What error do you get?
Check the corresponding entry in the "Troubleshooting" table on the ArduinoFDC GitHub page.

ACrazyTown, 2023-03-06:
> On the ArduDOS command line, type "monitor" which will get you into the low-level disk monitor. The disk drive motor should automatically turn on at that point. Check if it's running. If not, type [...]

Yup, the motor spins. I couldn't get the command you said working (said something about invalid sector specification) but typing just "r" gave me an error "Error: No sync marks found!". I'm going to try playing around with the GND pins and also a few other floppies. Thanks for your help!

closed 2023-04-14 by dhansel