dhansel/ArduinoFDC #11: Problem with changing the floppy disk.

GSoftwareDevelopment, 2023-09-21:
Hello.

By way of introduction - great project. I'm just playing with it and I'm going to use it as an interface for Atari 8-bit :) however...

The problem is that changing the floppy disk is not checked. It appears as follows:
1. I start the project.
2. I insert a floppy disk
3. I do for example `dir` - it shows me that it is empty and the amount of free space 
4. I change the floppy disk to another one 
5. I do `dir` and again it shows that it is empty and the same amount of free space - it is already suspicious. 
6. I do a reset of Arduino 
7. The same floppy disk as in point 4 
8. I do `dir` and it shows the contents of the floppy disk - some files.

GSoftwareDevelopment, 2023-09-21:
Thank you for your quick reply :) and my apologies, I must not have read.

Myślę, że można zamknąć temat. Pozdrawiam.

GSoftwareDevelopment, 2023-09-21:
Or not. :)

What is the reason for the lack of detection?

dhansel, 2023-09-22:
Laziness :) I was focused on reading/writing the actual data, which is what the ArduinoFDC library is about.
Adding ArduDOS was just an idea to demonstrate what can be done using the library.

Checking whether the disk has changed is not hard. Just connect pin 33 of the floppy cable to an unused
Arduino pin. In the sketch, call `pinMode(pin, INPUT_PULLUP)` to configure the input pin. Then you can
use `if( digitalRead(pin)==LOW ) { /* disk change detected */ }` to detect the change.

Note that the pin will **continue** reading LOW until you clear the "disk change" flag in the drive.
The only way to clear that flag is to move the disk drive head. So you'll have to (attempt to) read some 
sector data on a track different from the one that the head is currently on.

GSoftwareDevelopment, 2023-09-22:
heh... I see, I asked because I thought it was something more complicated :)

By the way, thanks for giving the way.

However... pin 33? isn't that GND by any chance? ;) 
I noticed, while looking for information about the DSKCHG# signal (i.e. pin 34), that - according to [Wiki](https://en.wikipedia.org/wiki/Floppy_disk_drive_interface) - it's an input pin, which surprises me a bit, that you set `pinMode()` to `INPUT_PULLUP` no but.... there's always a chance that there's an error in the Wiki :)

dhansel, 2023-09-22:
Oops sorry yes I meant pin 34. I fixed it in my original comment so as to not misinform anybody reading this thread. Thanks for the correction!

closed 2023-09-25 by dhansel