dhansel/ArduinoFDC #12: Very slow xmodem transfer, errors

slayernn021, 2023-09-27:
Hello! Im trying to copy files from old floppys, 3.5 using arduino mega, psu as ps. When i start receiving file via xmodem, it takes ~10min for 1kb and it gives lots of errors. Same with other floppy disks. I try using pullup resistor, and its same. All GND's are correct. What can be problem and where to look? Is there other way to copy files to computer except xmodem? Thanx!!

dhansel, 2023-09-27:
First thing to figure out is whether the problem is with XModem or reading the data itself. What do you get when you execute the "r" command (read whole disk and print pass/fail for each sector). Do you get lots of read errors? If so, try using an empty disk, format it and then try the "r" command. If that works then the other disk is likely to old/degraded to read using ArduinoFDC. If it still doesn't work then maybe the disk drive you are using has problems. 

If the "r" command works without issues then the problem is likely your serial connection between the Arduino and the PC. What terminal program do you use for the transfer? I've had good luck with TeraTerm.

Note that ArduinoFDC is not meant to archive old disks. It's purpose was to use an Arduino UNO to write and read data on floppy disks. ArduinoFDC _can_ be used for archiving if your disks and drive are in good condition. But it is not very good at error correction (because of the limited resources like speed/memory of the Arduino UNO). If you really want to archive disks I recommend using Greaseweazle/FluxEngine which read low-level data from a floppy and then use the power of a PC to interpret the data and correct possible errors.

thstrauss, 2023-10-02:
I had an issue with XModem transfer in my attempt do write a program which saves the content of an whole disk. Is solved it with using the implementation that is part of arduinoFDC on the PC. TeraTerm  also works perfect for me.

closed 2023-10-03 by dhansel