dhansel/ArduinoFDC #13: Problem when compile the code

thiago1255, 2023-11-08:
When i try compile the code i get this:
```
Arduino: 1.8.19 (Linux), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/internet/.arduino15/packages -tools /usr/share/arduino/hardware/tools/avr -tools /home/internet/.arduino15/packages -libraries /home/internet/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10819 -build-path /tmp/arduino_build_548072 -warnings=none -build-cache /tmp/arduino_cache_167950 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/internet/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/internet/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/home/internet/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/internet/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/home/internet/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/internet/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /home/internet/Documentos/ArduinoFDC/ArduinoFDC.ino
arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/internet/.arduino15/packages -tools /usr/share/arduino/hardware/tools/avr -tools /home/internet/.arduino15/packages -libraries /home/internet/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10819 -build-path /tmp/arduino_build_548072 -warnings=none -build-cache /tmp/arduino_cache_167950 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/internet/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/internet/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/home/internet/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/internet/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/home/internet/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/internet/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /home/internet/Documentos/ArduinoFDC/ArduinoFDC.ino
Using board 'mega' from platform in folder: /home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6
Using core 'arduino' from platform in folder: /home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated compiler.path={runtime.tools.avr-gcc.path}/bin/, automatically converted to compiler.path=/usr/bin/. Consider upgrading this core.
Detecting libraries used...
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino" "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/mega" "/tmp/arduino_build_548072/sketch/ArduinoFDC.ino.cpp" -o "/dev/null"
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino" "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/mega" "/tmp/arduino_build_548072/sketch/ArduinoFDC.cpp" -o "/dev/null"
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino" "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/mega" "/tmp/arduino_build_548072/sketch/XModem.cpp" -o "/dev/null"
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino" "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/mega" "/tmp/arduino_build_548072/sketch/diskio.cpp" -o "/dev/null"
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino" "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/mega" "/tmp/arduino_build_548072/sketch/ff.c" -o "/dev/null"
Generating function prototypes...
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino" "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/mega" "/tmp/arduino_build_548072/sketch/ArduinoFDC.ino.cpp" -o "/tmp/arduino_build_548072/preproc/ctags_target_for_gcc_minus_e.cpp"
"/usr/bin/arduino-ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_548072/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/usr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino" "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/mega" "/tmp/arduino_build_548072/sketch/ff.c" -o "/tmp/arduino_build_548072/sketch/ff.c.o"
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino" "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/mega" "/tmp/arduino_build_548072/sketch/ArduinoFDC.cpp" -o "/tmp/arduino_build_548072/sketch/ArduinoFDC.cpp.o"
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino" "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/mega" "/tmp/arduino_build_548072/sketch/ArduinoFDC.ino.cpp" -o "/tmp/arduino_build_548072/sketch/ArduinoFDC.ino.cpp.o"
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino" "-I/home/internet/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/mega" "/tmp/arduino_build_548072/sketch/XModem.cpp" -o "/tmp/arduino_build_548072/sketch/XModem.cpp.o"
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=1081 [... cut ...]
/tmp/ccZHqZaS.s:470: Error: constant value required
/tmp/ccZHqZaS.s:475: Error: constant value required
/tmp/ccZHqZaS.s:475: Error: constant value required
/tmp/ccZHqZaS.s:488: Error: constant value required
/tmp/ccZHqZaS.s:488: Error: constant value required
lto-wrapper: fatal error: /usr/bin/avr-gcc returned 1 exit status
compilation terminated.
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Mega or Mega 2560.
```

dhansel, 2023-11-08:
It looks like Arduino IDE version 1.8.19 on Linux uses an older gcc (version 5.4.0) which appears to have issues with the way ASM macros are used in ArduinoFDC.cpp.

I recommend using the new Arduino IDE version 2.2.1 which comes with a newer gcc (version 7.3.0) and can compile ArduinoFDC without errors.

On Windows both versions of the Arduino IDE com with gcc 7.3.0 so there is no issue on that platform.

thiago1255, 2023-11-09:
Use the new IDE make it works ! thank you !

closed 2023-11-09 by thiago1255