dhansel/ArduinoFDC #14: problem

popunia1, 2024-04-09:
Hi, I have a problem. I decided to connect my 5.25 floppy disk drive to the computer in this way, connected all the wires, connected the adruino nano to the computer, started downloading the sketch and an error appeared: "

avr-g++: error: {build.define0}: No such file or directory
avr-g++: error: {build.define1}: No such file or directory
avr-gcc: error: {build.define0}: No such file or directory
avr-gcc: error: {build.define1}: No such file or directory
avr-g++: error: {build.define0}: No such file or directory
avr-g++: error: {build.define1}: No such file or directory
avr-g++: error: {build.define0}: No such file or directory
avr-g++: error: {build.define1}: No such file or directory
avr-g++: error: {build.define0}: No such file or directory
avr-g++: error: {build.define1}: No such file or directory

exit status 1

Compilation error: exit status 1

". I'm not good at this and I'd appreciate your help.

dhansel, 2024-04-10:
Does the "blink" example compile? (Select File->Examples->01. Basics->Blink, then try to compile).
Also make sure to select the Arduino Nano as the board (in the "Tools" menu).

I have never seen these errors before. It looks like something is wrong with your Arduino IDE installation.
Are you on Windows? If so, I would recommend re-installing the Arduino IDE.

popunia1, 2024-04-10:
as a result, I installed the old version of IDE and everything worked out, thanks

closed 2024-04-10 by popunia1