dhansel/ArduinoFDC #15: Migtation to stm32

TEN1HC, 2024-06-23:
[FDD_Stm32_v03.zip](https://github.com/user-attachments/files/15944252/FDD_Stm32_v03.zip)
Im trying to migrate this lib to Stm32f103c8t6.
Using timer1 for input capture
Using timer3 for some index/pusle ..,check
Using timer2 for pwm out write data (not implemented yet)

Failure detected as f_open method => DISK_ERROR

So many thing to do and it still not work.
Could you help to check?

dhansel, 2024-06-23:
I had a look at your low-level reading code (function read_data in FDD.cpp), and it looks you don't quite have
a good grasp yet of how the MFM encoding on floppy disks works. It's not as simple as "short time between pulses=0"
and "long time=1". The read_data() function in my code does quite a bit more than that (but of course it's very specific
to the Atmega328p and therefore can't be used for STM32). I recommend reading up on the basics of MFM encoding
and how to read floppy disks. There's some good information here:
https://www.5volts.ch/posts/mfmreader
https://www.hermannseib.com/documents/floppy.pdf

Apart from that you have to be sure to get your timing right. There may only be 2us of time between two
pulses received from the drive. I don't know anything about STM32 but you may want to double-check
the interrupt latency since your CaptureCallback function relies on interrupts. If the latency is too high you
may miss some pulses. Also make sure that other interrupts don't get in the way,

This is definitely not an easy project to port to a different platform. It took me months of working with
an Oscilloscope and Logic Analyzer to get the timings right on the Atmega328p. I would suggest testing
at a lower level than the f_open() call. When I developed ArduinoFDC I started with the low-level disk monitor
that allowed me to just read (and write) data to one sector. I started with a known-good formatted disk
and used a program such as OmniFlop on a PC to get the expected content of a sector. Then on ArduinoFDC
read that same sector to see what I got. Use an Oscilloscope to really look at the timing, the points when
you read the data bits etc. Debug, rinse and repeat. It's a long process but very rewarding once you finally
get it right (and you'll learn a lot about debugging tools and techniques along the way).

Good luck!

TEN1HC, 2024-06-23:
Great to receive reply from you.
I realize that Im so naive when just give a simple pulse read method like that.
Thank you very much for the reference documentation.