dhansel/ArduinoFDC #19: Error: drive not ready and motor continually spinning when a floppy disk is inserted

Xcold0, 2025-07-18:
When I put a floppy disk in the drive and try to format or read a floppy disk the error drive not ready comes up. I've checked the wiring and that is correct and I've tried a different floppy disk the problem is still there. 

Also the error drive not ready also appears when I try to do anything with the disk when on is not inserted which has made me think this might be an issue with the drive itself, though I am not sure.

This is on a 3.5 HD drive. The disks used are unformatted disks.

closed 2025-07-19 by Xcold0

montyleib, 2025-07-19:
Assuming that you found your problem, would you like to share your solution in case someone else runs into it?

Xcold0, 2025-07-20:
It was improper grounding that was the issue one of the GND pins didn't go in all the way